iboguslavsky/pwm-sunxi-opi0 #1: PWM1

alexusvm, 2017-04-10:
Hello, iboguslavsky!
**Thanks for your pwm-sunxi-opi0 driver**. It's work with servo! Can you rebuild this driver for work with PWM1 (PA06) on Orange Pi Zero?
Thank, Alexey.
PS. Sorry for my English.

iboguslavsky, 2017-04-11:
Alexey,

This will be added once Allwinner publishes H2+ datasheet. The only documented SoC that is close to it (H3) has only one PWM channel defined. We need this info to be able to address PWM1's registers.

Igor

closed 2017-04-11 by iboguslavsky

alexusvm, 2017-04-11:
http://dl.linux-sunxi.org/A33/A33%20user%20manual%20release%201.1.pdf - not this?

iboguslavsky, 2017-04-14:
Thanks Alexey. I have given it a try. Close, but no cigar - the PWM1 on PA6
still remains a mystery. I have re-factored the driver code to accommodate
multiple PWMs (see "twopwms" branch). Using pin config data from A33 still
didn't produce a working PWM on PA06. Let me know when you get ahold of H2+
datasheet, and will accommodate this quickly.