iboguslavsky/pwm-sunxi-opi0 #2: servo motor continuously rotating

pinootto, 2017-05-07:
Hello,
I tried to run the pwm-sunxi-opi0 on my orange pi plus 2e.
It compiles and I can insert the module into the kernel successfully.
Then I set up the various parameters successfully. But when I run the pwm, the servo motor (ES08MAII) starts to spin forever. It does not put itself in a position between 0 and 180 degres, but it rotates continuously.
I also noticed that after I stop the run, the parameters that I set up previously are lost. I have to set them all again.
Any idea about how to fix this problem?
Best regards,
Giovanni

iboguslavsky, 2017-05-07:
What values are you using for prescale / active  / entire cycles? Stopping PWM does clear out the settings - try using the version of the driver under https://github.com/iboguslavsky/pwm-sunxi-opi0/tree/twopwms

pinootto, 2017-05-08:
Thanks for your answer.
Now I succeeded to use it. First I have to give the run command and then the settings. Before I was doing the opposite: first the settings and then the run.
I think that you could write a note on the README about the correct sequence to set the values and the run, so other people will not have the same problem.
You can also add that it works on Orange Pi Plus 2E.
Thanks again for your efforts and your sharing.
Best regards,
Giovanni

iboguslavsky, 2017-05-08:
Done - thanks for the comment.

pinootto, 2017-05-08:
You are welcome.

closed 2017-05-09 by iboguslavsky