iboguslavsky/pwm-sunxi-opi0 #3: Problems under pwm-sunxi-opi0 driver

andreipop106, 2017-05-08:
Hello,

First of all thank you for providing this PWM driver for OPI Zero.
As far as I can see tree folder must be created at every start up (I do not understand why).

I can NOT generate PWM on any pin. I have creates specific shell scripts as stated by you.
As far as I can see OPI has pins configuration of PWM1 as in the link.
LINK: https://www.google.ro/search?q=orange+pi+pin+layout&safe=active&rlz=1C1CHBF_enRO697RO697&tbm=isch&imgil=RqsPbXBPIKoYIM%253A%253Bx5N83oJxGk8icM%253Bhttps%25253A%25252F%25252Foshlab.com%25252Forange-pi-zero-pinout%25252F&source=iu&pf=m&fir=RqsPbXBPIKoYIM%253A%252Cx5N83oJxGk8icM%252C_&usg=__8HjGMDL7BV6T4_sKWV2Fn4IcVd4%3D&biw=1455&bih=731&ved=0ahUKEwje4KPUzNvTAhWKvRoKHVegDUwQyjcIPg&ei=2e0NWZ6QEor7atfAtuAE#imgrc=RqsPbXBPIKoYIM:
I have tried configuring script.bin for both PA5 and PA6 and is not working for me.
All scripts are called with success, tried from command line also but with no success.

Please help me out!
Lucian Pop

iboguslavsky, 2017-05-08:
Lucian - the tree is created in the sysfs filesystem, which is memory - mapped; it gets wiped out every time you reboot. You may want to consider the following:

Copy my kernel module to the drivers directory.
$ sudo cp mymodule.ko /lib/modules/$(uname -r)/kernel/drivers/

Add the module to the file /etc/modules. 
$ echo pwm-sunxi-opi0 | sudo tee -a /etc/modules

Update the list of module dependencies.
$ sudo depmod

iboguslavsky, 2017-05-08:
In my experiments, PA5 (PWM0) is the only PWM that works on OPI0. it's the one on Rx pin on the board's UART.

andreipop106, 2017-05-09:
Hello Iboguslavsky,
Thank you for such short notice, I am trying to port application from RaspberryPi2 --> OrangePiZero.I am glad that someone is interested in this OPI board.
Is there a way of controlling this PWM0 via C, without using shell scripts to do so? (maybe including your .c file provided by you or somehow?)I am not sure I got correctly the PA5 location. I have found the PA5/PWM0/UART0_RX inside datasheet but only near CPU pins. Did I get it correctly?

Thanks in advance for all your effort. :)
 

    În Luni,

iboguslavsky, 2017-05-09:
Lucian, 

PA5 is located on the UART header. It's a multi-functional pin. PA5 is the one in the middle of the header (grey color, see picture below). 

https://github.com/iboguslavsky/pwm-sunxi-opi0/blob/master/images/pwm0.jpg

iboguslavsky, 2017-05-09:
...and yes, you can certainly use C (or any other conceivable language) to develop apps against this driver. It comes down to opening files in sysfs (/sys/class/pwm-sunxi-opi0/pwn0/entire_cycles, active_cycles, prescale, etc.) and writing values to them. Let me know if you need help.

andreipop106, 2017-05-25:
Hello again,

I have create C handling application for provided driver by you.
One small question:
Why "PRESCALE_INVx05  = 0x05, // Invalid prescaler setting" is this so?
I am wanting to control DC motors (like water pump or brushed DC motor both 12VDC), as I have understood 25kHz via 40 us is the best design for doing so.

Can you please provide me private email?

Thanks a lot for the knoledge. :)
Best regards,
Lucian Pop