iboguslavsky/pwm-sunxi-opi0 #4: Editing prescale values

mtanwr, 2017-05-15:
Thank you @iboguslavsky  for creating this repo for OPI0.

I am having trouble understanding how to edit the prescale values. For example, if I want to specify PRESCALE_DIV_NO, then do I write the hex value 0x0f in prescale file? Or do I need to convert that to some other format? Like decimal, which in this case would be 15. 

When I try to write 0x0f, then it becomes 0 and saves it like that. It seems it only takes decimal values of 0 to 15 as valid arguments.

iboguslavsky, 2017-05-15:
Assuming you first write "1" to "run" file (you have to do it before you write any values to "prescale" file) - then yes, you just write a numeric value into "prescale" file (and not a string representation of it, like "0x0f"). You can, indeed, use decimal representation (15), hex (0x0f), or any other - as long as they are numeric values, not strings,

Share your your code sample here if you need additional help.

mtanwr, 2017-05-16:
Thanks! I got the pwm running with no problems. I don't have any code yet, as I was just testing OPI pwm through command line, but I'll share if I have any further problems.

closed 2017-05-16 by mtanwr