iboguslavsky/pwm-sunxi-opi0 #5: Make fails

Hengy, 2018-06-01:
I know it's been a few years, however, I just tried to install this on a Orange Pi Zero on Armbian 3.4.113, and I get this error:
`make -C /lib/modules/3.4.113-sun8i/build/ M=/home/reflow/pwm-sunxi-opi0 modules
make[1]: Entering directory '/lib/modules/3.4.113-sun8i/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/3.4.113-sun8i/build'
Makefile:4: recipe for target 'all' failed
make: *** [all] Error 2
`

I tried installing linux headers with
`apt-get install linux-headers-$(uname -r)`
but that didn't solve it.

iboguslavsky, 2018-06-07:
I think you might need the entire kernel source, not just headers...

ELDJAZAERY, 2019-06-30:
@Hengy did you find a solution to your problem?

iboguslavsky, 2019-06-30:
You will likely also need you kernel sources (on top of just headers) either
from the*kernel.org <http://kernel.org>* network or your distributor's
kernel source package.

ELDJAZAERY, 2019-07-04:
> kernel sources (on top of just headers)

Sir, I have no idea how to use the kernel and the headers to build this code, I am using orange pi one with ARMBIAN 5.83 stable Ubuntu 16.04.6 LTS 3.4.113-sun8i.
if you could guide me how to do it...! I'd really appreciate your help.

joek85, 2019-07-05:
You need to compile it inside the kernel source tree, just copy the "pwm-sunxi-opi0.c" to "linux kernel source tree/drivers/pwm" then add "obj-m+=pwm-sunxi-opi0.o" to the make file inside that folder and compile the kernel, you should see the ko file in the modules directory. Hope this helps.