iboguslavsky/pwm-sunxi-opi0 #6: Linux 4.14 build fixes (not a issue the real fixes)

fgl27, 2018-12-16:
I adapt the driver to build using latest armbian https://github.com/armbian/build

Just sharing

https://github.com/fgl27/pwm-sunxi-opi0/commit/24759bffc5c560b4294be8d234f8dbe83d6c7944

Also in a  patch version

https://github.com/fgl27/pwm-sunxi-opi0/blob/master/patchs/pwm_sunxi_linux_4_14.patch

Works today will probably work for some time...

I also made a simple table as a help I believed is OK, because I did not had time to test under a oscilloscope and will not for some time.

**Sample of frequency values**

|prescale (in decimal) | entire_cycles | Result in Hz (Prefix Hz) | active_cycles (Duty%)|
|-------|-------|----------------------|--------|
|15     | 1     | 12000000 Hz (12 MHz) | 1 (50%)|
|15     | 10    | 2400000 Hz (2.4 MHz) | 5 (50%)|
|15     | 100   | 240000 Hz (240 kHz)  | 50 (50%)|
|15     | 1000  | 24000 Hz (24 kHz)    | 500 (50%)|
|15     | 10000 | 2400 Hz (2.4 kHz)    | 5000 (50%)|
|0      | 1     | 100000 Hz (100 kHz)  | 1 (50%)|
|0      | 10    | 20000 Hz (20 kHz)    | 5 (50%)|
|0      | 100   | 2000 Hz (2 kHz)      | 50 (50%)|
|0      | 1000  | 200 Hz (0.2 kHz)     | 500 (50%)|
|0      | 10000 | 20 Hz (0.02 kHz)     | 5000 (50%)|

nice work @iboguslavsky thanks, close this if you wanna :+1:

iboguslavsky, 2018-12-18:
> fgl27@24759bf

Awesome job, Felipe! Can you open a pull request so i could merge it?

Thanks,
Igor

fgl27, 2018-12-19:
Shore no problem... I just made two, one with the fix for 4.14, and a separated with a readme update just in case you don't wanna the readme changes.

A god idea is to make a legacy branch before merging the PR.

iboguslavsky, 2018-12-19:
Yep - I’ll make 2 branches - one for legacy kernel, and one for mainstream.

Thanks again and good luck with your project,
Igor